ronoth/LoStik #2: expand README

vielmetti, 2018-07-04:
Some things I'd expect to see in a README

- what this code is for
- who wrote it
- where you can order these parts
- when things are expected to ship
- why you might want it
- what else you might build on top of this system

vielmetti, 2018-07-04:
Some of the details and pictures from https://www.thethingsnetwork.org/forum/t/lora-stik-rn2903-usb-device/14976 would be good to add here too.

vielmetti, 2018-07-04:
Some reference to the RN2903 data sheet at http://ww1.microchip.com/downloads/en/DeviceDoc/50002390E.pdf would be useful too.

lolsborn, 2018-07-09:
Thanks @vielmetti  I have plans to add more example code and an updated read me in the following weeks before the CS campaign closes.

WA4OSH, 2018-12-01:
Some reference to the RN2483 data sheet would also be good for EU868 users.

WA4OSH, 2018-12-01:
Some reference to the PIC18F46K22 is needed
https://www.microchip.com/wwwproducts/en/PIC18F46K22
http://ww1.microchip.com/downloads/en/DeviceDoc/40001412G.pdf

WA4OSH, 2018-12-01:
Some reference to the LoRaWAN Library is needed:
http://ww1.microchip.com/downloads/en/DeviceDoc/40001864B.pdf

WA4OSH, 2018-12-01:
Your LoStik API needs to be documented
  send_cmd(" ")...

  sys set pindig GPIO11 0    
      GPIO10 = blue led
      GPIO11 = red led
      0 = off?
      1 = on?

  sys get ver

  radio get freq
  radio get mod
  radio get sf

  radio rx

  radio set pwr 10

...etc.

Where are the LoRaWAN commands?  Are there callbacks to handle the incoming LoRaWAN messages?

lolsborn, 2018-12-01:
Check out the command references linked to in the readme.

WA4OSH, 2018-12-05:
@lolsborn  OK ... then please document at least the action of the system commands that won't be documented in the Microdot document.  For example, there's no indication in the microdot documentation as to which LEDs are connected to what ports.  Yes, I can read a schematic.  But, it's sometimes good to have it all in one place.  Documentation does not need to be a research project for the user.

sys set pindig 
GPIO10 = blue led
GPIO11 = red led
0 = off?
1 = on?

lolsborn, 2018-12-08:
> Need a quick "sanity check" for LoStik.
> Configure Tera Term for Serial port, 56700 baud, transmit CR_LF,
> receive Auto, Local echo, etc.>  Then a couple of lines of text that verify that LoStik is running.

Try just sending "sys get ver" to start.  If that doesn't return a
version string then you then you may have an issue with your unit.Sending random text will also return "invalid_command".    Otherwise
your terminal settings look correct.

Links:

  1. https://github.com/lolsborn/LoStik/issues/2#issuecomment-445459253
  2. https://github.com/

lolsborn, 2018-12-08:
> It did not respond to the sys get ver until I talked to it with the
> LoRa Development Suite.  I did  successfully configure it for my
> RAK831 gateway using OTAA.  I was able to join it.  My next steps are
> to get it to send a fake temperature humidity and GPS location message
> that's Cayenne formatted using a Python script.  I will post that

There is a miniterm.py in examples that works well as a test terminal.
If you run python miniter.py <COMPORT> 57600 it is a good way to see if
it's an issue with your existing terminal settings.

Links:

  1. https://github.com/lolsborn/LoStik/issues/2#issuecomment-445475790
  2. https://github.com/

WA4OSH, 2018-12-08:
I had not setup Python 3.7.1 on my machine yet and needed something quicker
to check my LoStik.  Next step, is of course to finish getting Python up
and running.

--Konrad

lolsborn, 2018-12-09:
Added info about LEDs to readme and have updated docs / examples quite a bit since this ticket was initially created.  I'm going to close it for now.  If there are specific suggestions or PRs please open another issue.

closed 2018-12-09 by lolsborn