ronoth/LoStik #3: radio_sender.py: invalid_param

dl8bb, 2018-11-09:
Hi folks,
  today I got the two LoStiks from customs and I'm happy to try it out. I cloned the repo and started trying the examples. blinky.py is working with both sticks but I have not much luck with radio_sender.py and radio_receiver.py scripts. It looks like the sender is only working the 1st time and than it says "*invalid_param*" forever.

Here is the output I see:

```
pi@pinebook:~/LoStik/examples$ ./radio_sender.py
ok
RN2483 1.0.4 Oct 12 2017 14:59:25
lora
868100000
sf12
4294967245
ok
ok
ok
ok
ok
radio_tx_ok
ok
invalid_param
ok
ok
invalid_param
ok
ok
invalid_param
ok
ok
invalid_param
^CTraceback (most recent call last):
  File "./radio_sender.py", line 26, in <module>
    time.sleep(2)
KeyboardInterrupt
pi@pinebook:~/LoStik/examples$
```

and the receiver is on a pi2

```
pi@raspberrypi:~/LoStik/examples $ ./radio_receiver.py
connection made
RN2483 1.0.4 Oct 12 2017 14:59:25
4294967245
radio_rx  1541778041
```

Any hints or help for me?

BTW: I modified the parser.add_arguments() call in both files to
```
parser.add_argument('port', nargs='?', default='/dev/ttyUSB0', help="Serial port descriptor")
```
cheers
birko
--

dl8bb, 2018-11-09:
I've added some debug output to the radio_sender.py script and it looks like at some point the LoStik doesn't understand the command *sys set pindig GPIO11 0* any more.
```
pi@raspberrypi:~/LoStik/examples $ ./radio_sender.py
### cmd: sys set pindig GPIO11 0
### response: ok
### cmd: sys get ver
### response: RN2483 1.0.4 Oct 12 2017 14:59:25
### cmd: radio get mod
### response: lora
### cmd: radio get freq
### response: 868100000
### cmd: radio get sf
### response: sf12
### cmd: mac pause
### response: 4294967245
### cmd: radio set pwr 10
### response: ok
### cmd: sys set pindig GPIO11 0
### response: ok
### cmd: sys set pindig GPIO11 1
### response: ok
### cmd: radio tx 1541779677
### response: ok
### cmd: sys set pindig GPIO11 0
### response: ok
### cmd: sys set pindig GPIO11 1
### response: radio_tx_ok
### cmd: radio tx 1541779679
### response: ok
### cmd: sys set pindig GPIO11 0
### response: invalid_param
### cmd: sys set pindig GPIO11 1
### response: ok
### cmd: radio tx 1541779681
### response: ok
### cmd: sys set pindig GPIO11 0
### response: invalid_param
^CTraceback (most recent call last):
  File "./radio_sender.py", line 27, in <module>
    time.sleep(2)
KeyboardInterrupt
pi@raspberrypi:~/LoStik/examples $
```
On the receiver side I only see the 1st transmission.
```
bergt@pinebook:~/LoStik/examples$ ./radio_receiver.py
connection made
RN2483 1.0.4 Oct 12 2017 14:59:25
4294967245
radio_rx  1541779677
```
I changed the sender and receiver to the other computer.

olemd, 2018-11-12:
Adding a slight delay after 
`send_cmd("sys set pindig GPIO11 1")`
"works for me".
`time.sleep(.1)`

Removing the whole "sys set pindig GPIO11 1" also works, presumably blinking the LED takes a moment and the stick isn't immediately ready for a new command.

dl8bb, 2018-11-13:
Removing both commands to switch the LED on and off helped. 
The short delay of .1 or .2 didn't work for me.
```
    while True:
        time.sleep(2)
#        send_cmd("sys set pindig GPIO11 1")
        send_cmd('radio tx %d' % int(time.time()))
        time.sleep(.1)
#        send_cmd("sys set pindig GPIO11 0")
#        time.sleep(.2)
```

olemd, 2018-11-13:
I suppose this could be related to CPU-speed. I've just tested it on my laptop with a i7-6500U. The .1 sleep delay worked fine there, but I suspect a RPi or other low-end system might not need it while a faster CPU might need an even higher delay. Assuming send_cmd and python itself will take a tiny bit longer to execute on the low-end systems?

liverun, 2018-11-24:
Adding a slight delay after
`send_cmd("sys set pindig GPIO11 1")`
worked for me:
`time.sleep(.1)`

The sender runs on a RPi

lolsborn, 2018-12-09:
Hey guys, somehow this whole thread slipped past me.  I'm going to update the example code to include an additional timeout.  I'm not sure why @dl8bb would need to comment out that code.  @dl8bb have you tried running those commands separately?  It's possible to wait for an "ok" before moving to the next command. I might play around with something like pexcept in python.

lolsborn, 2018-12-10:
I think this commit will fix the issue as well as blink the led on send properly and it also sends a frame counter which makes it easier to catch missed frames on the receive.

closed 2018-12-10 by lolsborn