ronoth/LoStik #5: PIC18LF46K22 code?

WA4OSH, 2018-12-01:
I don't see the PIC18LF46K22 code in the GitHub.  Does it have API primitives to support LoRaWAN as well as LoRa?

WA4OSH, 2018-12-01:
Please document the API to the PIC18LF46K22 code so that we can develop LoRa (point-to-point) and LoRaWAN connections.

WA4OSH, 2018-12-01:
Does your PIC's LoRaWAN code support OTAA, or is it ABP only?

lolsborn, 2018-12-09:
The default firmware on the RN2903 is UART / text based.  See command reference linked in readme.  It supports both OTAA and ABP class A LoRaWAN.

I'm not going to include any instructions on here for programming via C / MPLab as 99% of people are using this as a test device via the UART interface.  I have had some success programming the PIC18 on the RN2903, but it requires a lot of effort to get setup.  The firmware library for the PIC includes both class A and class C LoRaWAN support.

closed 2018-12-09 by lolsborn

magillus, 2019-06-13:
I found that firmware of RN2903 1.0.5 at https://www.microchip.com/wwwproducts/en/RN2903 release notes mention class C support with `set class` command