ronoth/LoStik #8: LoStik is not recognised on OSX HighSierra

krital, 2019-01-10:
Hi,

do I need some special FTDI / other driver for LoStik to be recognised under OSX? I have tried both with and without a hub and I get no serial device created.

lolsborn, 2019-01-11:
Could you paste the output of 

    system_profiler SPUSBDataType	

We're looking for a device that matches Product ID: 0x7523 and Vendor ID: 0x1a86 if we have that it's probably not a hardware issue at least.

Once we know it's not a hardware issue take a look in /dev/tty.*  for some reason the ch340 on my mac shows two devices a tty.usbserial- and tty.wchusbserial  the later is the one you want to connect to.  If you don't see a device there I recommend taking these steps:

https://medium.com/@tony92151/arduino-ch340-chip-driver-for-high-sierra-39e621dae1e4

The terminal requires CRLF which is your terminal app may or may not use by default.  I would recommend starting off using the miniterm.py script in examples mentioned in the readme.

krital, 2019-01-12:
Hello,

its not a hardware issue, I can see the device:

   USB2.0-Serial:

          Product ID: 0x7523
          Vendor ID: 0x1a86
          Version: 2.63
          Speed: Up to 12 Mb/sec
          Location ID: 0x00200000 / 13
          Current Available (mA): 500
          Extra Operating Current (mA): 0

I can also see it with lsusb: Bus 000 Device 013: ID 1a86:7523 1a86 USB2.0-Serial

What is do not see is a serial device under /dev/tty.* hence I can not run any terminal... I did try with minicom initially against /dev/ttyS0 but that is not the correct serial port. I have faced the problem in the past with various boards and I had to install some FTDI drivers.. I understand though that LoStik uses some other chip?

krital, 2019-01-25:
hello, so as the device does not appear, I followed the steps on the medium post above, rebooted my Mac as instructed but I still do not see the device. What do I try next?