ronoth/LoStik #11: LoStik: needs driver in raspberry PI?

neypalma, 2019-03-25:
Hi, i want to use the lostik with my raspberry 3. The raspberry seems to recognize the lostik but i cant connect to serial port. 

Mi question is if lostik needs driver installation in raspberry.

Greetings

lolsborn, 2019-03-25:
What distro are you using?

neypalma, 2019-03-25:
> What distro are you using? 
Im using rasbian stretch

lolsborn, 2019-03-25:
So I installed rasbian-strech-full on SD card, booted it and the device mounted to /dev/ttyUSB0

Unfortunately I didn't get any further before my RPI A croaked. I'm not sure what happened, but it's no longer with us. 

Are you not seeing a device at /dev/ttyUSB0

neypalma, 2019-03-25:
Hi, yes I see TTYUSB0, I connect to it but can interact with the board. There is no message. The device seems to be hanged after making the connection

lolsborn, 2019-03-25:
What terminal program are you using to connect?

neypalma, 2019-03-25:
I try putty and minicom

lolsborn, 2019-03-25:
Putty is known to have issues because the RN2903 requires \r\n endings.

I'm not really familiar with minicom. I would recommend trying the miniterm.py script in the examples directory first. There is also a gui program released by Microchip, but requires Java to run.

There is a bit more info on how to run miniterm and where to download the LoRa Development Suite Gui here.
https://github.com/ronoth/LoStik

neypalma, 2019-03-26:
@lolsborn  i have had success using picocom program and running it like this

 

> picocom --omap crcrlf,delbs --imap crcrlf,delbs -c -b 57600 -r -l /dev/ttyUSB0

Greetings

closed 2019-03-31 by neypalma