ronoth/LoStik #12: Problem connecting module via otaa

neypalma, 2019-04-02:
Hi, im trying to connect to a ttn app on a gateway. Im using OTAA join but the module always send me a denied response.

Anyone had succeeded connecting the module to a ttn app via OTAA?

Greetings

lolsborn, 2019-04-02:
There are quite a few of us using LoStik with TTN, I've connected via OTAA and ABP

Are you trying to connect manually or are you running the python example?

If you are doing it manually could you supply the commands (minus credentials) that you are sending?

What is the exact error you are seeing? Is it timing out or being actively denied?

neypalma, 2019-04-02:
Hi, im trying to connect manually. These are the commands that i have issued (the problem is actively denied)

```
mac set deveui
mac set appeui
mac set appkey
mac set adr on
mac join otaa
```

Greetings

lolsborn, 2019-04-02:
Are you using EU or US version?

> Hi, im trying to connect manually. These are the commands that i have issued (the problem is actively denied)

> `mac set deveui
mac set appeui
mac set appkey
mac set adr on
mac join otaa
`

neypalma, 2019-04-02:
> Are you using EU or US version?
Im using US version (915Mhz)

lolsborn, 2019-04-02:
Hrm,

Your commands look correct. I know some people using the EU version have reported they had to do

mac pause
mac resume

for some reason before configuring the mac

This is one contributed code example which is just sending the same commands via python

https://github.com/ronoth/LoStik/blob/4b109759486caca33381d06ae525d0ba2a6854f1/examples/lorawan_sender.py

And a very similar example included in the examples directory

https://github.com/ronoth/LoStik/blob/master/examples/lorawan.py

Both of them are sending the same basic commands that you are. The only reason I can think of TTN actively rejecting would be invalid / wrong credentials.

I can test it again manually this afternoon and try to do a screen capture if that would be helpful.

neypalma, 2019-04-02:
> I can test it again manually this afternoon and try to do a screen capture if that would be helpful.

I will appreciate the help. I will try with the links you gave me.

Greetings

neypalma, 2019-04-06:
Hi @lolsborn can you try to connect manually?

lolsborn, 2019-04-08:
What gateway are you using?

My guess is you are using a single channel or 8 channel gateway and the channel map on the RN2903 has all 72 channels enabled.  This way it tries to connect by picking a channel at random and chances are it picks one that is not open on your gateway.

Try running the commands below to configure your channels like this:

```
mac set ch status 0 off
mac set ch status 1 off
mac set ch status 2 off
mac set ch status 3 off
mac set ch status 4 off
mac set ch status 5 off
mac set ch status 6 off
mac set ch status 7 off
mac set ch status 8 on
mac set ch drrange 8 0 3
mac set ch status 9 on
mac set ch drrange 9 0 3
mac set ch status 10 on
mac set ch drrange 10 0 3
mac set ch status 11 on
mac set ch drrange 11 0 3
mac set ch status 12 on
mac set ch drrange 12 0 3
mac set ch status 13 on
mac set ch drrange 13 0 3
mac set ch status 14 on
mac set ch drrange 14 0 3
mac set ch status 15 on
mac set ch drrange 15 0 3
mac set ch status 16 off
mac set ch status 17 off
mac set ch status 18 off
mac set ch status 19 off
mac set ch status 20 off
mac set ch status 21 off
mac set ch status 22 off
mac set ch status 23 off
mac set ch status 24 off
mac set ch status 25 off
mac set ch status 26 off
mac set ch status 27 off
mac set ch status 28 off
mac set ch status 29 off
mac set ch status 30 off
mac set ch status 31 off
mac set ch status 32 off
mac set ch status 33 off
mac set ch status 34 off
mac set ch status 35 off
mac set ch status 36 off
mac set ch status 37 off
mac set ch status 38 off
mac set ch status 39 off
mac set ch status 40 off
mac set ch status 41 off
mac set ch status 42 off
mac set ch status 43 off
mac set ch status 44 off
mac set ch status 45 off
mac set ch status 46 off
mac set ch status 47 off
mac set ch status 48 off
mac set ch status 49 off
mac set ch status 50 off
mac set ch status 51 off
mac set ch status 52 off
mac set ch status 53 off
mac set ch status 54 off
mac set ch status 55 off
mac set ch status 56 off
mac set ch status 57 off
mac set ch status 58 off
mac set ch status 59 off
mac set ch status 60 off
mac set ch status 61 off
mac set ch status 62 off
mac set ch status 63 off
mac set ch status 64 off
mac set ch status 65 on
mac set ch status 66 off
mac set ch status 67 off
mac set ch status 68 off
mac set ch status 69 off
mac set ch status 70 off
mac set ch status 71 off
```

neypalma, 2019-04-08:
@lolsborn i will try what you comment. 

Greetings

neypalma, 2019-04-10:
Dear @lolsborn 

I have tried the commands without success. Im able to connect via abp, but the app seens to never receive the mac tx uncnf. (it has the never seen status). Im using a single channel gateway made with lopy board with to external antennas (one for LoRa, one for WIFI). Any ideas?

Greetings

lolsborn, 2019-04-10:
Something like this?
https://www.hackster.io/bucknalla/lopy-lorawan-nano-gateway-using-micropython-and-ttn-a9fb19

neypalma, 2019-04-10:
@lolsborn yes thats the gateway im using

lolsborn, 2019-04-10:
What is your frequency and LORA_DR set to in config

neypalma, 2019-04-10:
@lolsborn this is the config used

`# for US915
LORA_FREQUENCY = 903900000

LORA_GW_DR = "SF10BW125" # DR_0

LORA_NODE_DR = 0`

lolsborn, 2019-04-10:
I ordered a LoPy to test, but I'm guessing what you want to do is disable all but the channel the LoPy is set to and adjust the DR for that channel accordingly on the LoStik.

neypalma, 2019-04-28:
Hi, im trying to connect via ABP but mc tx cnf always returns mac_err. Have you try with a lopy?

Greetings

lolsborn, 2019-04-29:
Have you configured the channels on LoStik to correspond to your LoPy setup? I haven't tried yet, still waiting on my lopy.

lolsborn, 2019-04-29:
Are you using the LoPy1 or LoPy4?

neypalma, 2019-04-29:
@lolsborn I'm using lopy4

lolsborn, 2019-04-29:
Ok, that is what I ordered. I should have one here on Wed or Thursday.

neypalma, 2019-04-29:
What is the configuration for channels that I have to do it? What's the basic configuration for connecting abp and send a confirmed package to ttn in my lopy nanogateway simple channel? Always give mac_err

hallard, 2020-07-28:
Experienced same issues it's a know issue with RN2483 and TTN with SF12. since I setup starting OTAA with SF7 (DR5) no problem anymore, just good to know.

haimiko, 2021-11-24:
> What gateway are you using?
>
> My guess is you are using a single channel or 8 channel gateway and the channel map on the RN2903 has all 72 channels enabled. This way it tries to connect by picking a [...]

What's the channel setting for 915 sub-band 0 or sub-band 1?

Moon-Shards, 2024-05-15:
> What gateway are you using?
>
> My guess is you are using a single channel or 8 channel gateway and the channel map on the RN2903 has all 72 channels enabled. This way it tries to connect by picking a [...]

I almost lost my mind trying to connect my module to my 8-channel LoRa Gateway. I manually turned off all the channels with `mac set ch status x off` where x are all the channels from 1 to 71, leaving only channel 0. 

```
mac join otaa
ok
accepted
```

It worked!