ronoth/LoStik #13: Does LoStik work with Android?

DimaBond174, 2019-04-23:
Hi

Can i connect LoStik device to Android phone and work with them through OTG?
Will it work for Android 8, 9 (API > 24)?
Is the power supply from the phone's OTG enough for  to work the LoStik?

Thanks in advance for the answers..

lolsborn, 2019-04-23:
I don’t know anyone who has tried with Android. It is a serial text device using a CH340 usb to serial converter. It works with other flavors of Linux including Raspian, but I don’t currently have an Android device to test on.

lolsborn, 2019-04-23:
This app claims to support FTDI and CH340G devices so it seems like it's a possibility. https://play.google.com/store/apps/details?id=com.liorhass.android.usbterminal.free&hl=en_US

closed 2019-04-29 by lolsborn