ronoth/LoStik #16: mRuby

zwykl3, 2019-05-03:
Is possible using mRuby on this device?

lolsborn, 2019-05-03:
Are you inquiring about LoStik or LoDev?

Also, I'm assuming you are talking about MicroRuby and not mRuby?

LoStik is a device that allows you to send LoRa messages by talking to it over UART and won't run custom code. It is possible to upload your own firmware to the on-board PIC micro, but I don't know anyone besides myself who has tried it.

LoDev is a micro-controller board using a LoRa/STM32 SoC and it maybe possible tot run MicroRuby on it. No one has attempted it to my knowledge. I have spent some time on MicroPython, but so far the tests haven't been super promising 192k flash is a bit tight for MicroPython. It may still be possible, but will require some non-trivial trimming.

zwykl3, 2019-05-05:
I'm interesting Lora communications. In my opinion internet isnt free.
mRuby need 400Kb memory to run. MicroRuby is not a oficial way to using Ruby. I like Ruby and write some extension in C. mRuby (and mrubyc) are created in C. Every procesor who have c compiler and  malloc can run mRuby.

Lostick for me are great way to connect people like a old BBS or services similar usenet or irc/communicators. 
For example how connect pygo, gotenna and Lostick? Idea need more flexible and more programmers

Meybe in Lostick 2.0 You put more RAM.

closed 2019-06-14 by lolsborn