ronoth/LoStik #18: No data received with example codes

m0nkeyc0de, 2019-06-10:
Hello, beginning of the year I bought a pair of 868MHz LoStik and now I'm trying to have the examples working.

It turns out that the "radio_receiver.py" doesn't receive any data. I modified it to print every call to "handle_line" and here is the output I get :
```
(LoEnv) thor@valhalla:~/Geeking/LoStik/examples$ ./radio_receiver.py /dev/ttyUSB1
connection made
handle_line(data='RN2483 1.0.4 Oct 12 2017 14:59:25')
RN2483 1.0.4 Oct 12 2017 14:59:25
handle_line(data='4294967245')
4294967245
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='busy')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='busy')
handle_line(data='radio_err')
handle_line(data='ok')
handle_line(data='radio_err')
handle_line(data='ok')
... and so on...
```
The sender seems to work be working fine :
```
(LoEnv) thor@valhalla:~/Geeking/LoStik/examples$ ./radio_sender.py /dev/ttyUSB0
radio tx 5cfea21d0
radio tx 5cfea21f1
radio tx 5cfea2212
radio tx 5cfea2243
radio tx 5cfea2264
radio tx 5cfea2285
radio tx 5cfea22a6
radio tx 5cfea22d7
radio tx 5cfea22f8
radio tx 5cfea2319
radio tx 5cfea234a
radio tx 5cfea236b
radio tx 5cfea238c
```

I tried to invert LoStik roles but it doesn't change anything. 
Any clue ?

lolsborn, 2019-06-13:
I was able to reproduce this issue this morning. It seems to only affect the EU version for some reason, not the US radios. I'm guessing more configuration is required on startup, but it has been tested with earlier versions of this code so I'm not sure what is going on. I'll continue to investigate and hopefully push a patch today.

> Hello, beginning of the year I bought a pair of 868MHz LoStik and now I'm trying to have the examples working.

> It turns out that the "radio_receiver.py" doesn't receive any data. I modified it to print every call to "handle_line" and here is the output I get :

> `(LoEnv) thor@valhalla:~/Geeking/LoStik/examples$ ./radio_receiver.py /dev/ttyUSB1
connection made
handle_line(data='RN2483 1.0.4 Oct 12 2017 14:59:25')
RN2483 1.0.4 Oct 12 2017 14:59:25
handle_line(data='4294967245')
4294967245
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='busy')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='busy')
handle_line(data='radio_err')
handle_line(data='ok')
handle_line(data='radio_err')
handle_line(data='ok')
... and so on...
`
> The sender seems to work be working fine :

> `(LoEnv) thor@valhalla:~/Geeking/LoStik/examples$ ./radio_sender.py /dev/ttyUSB0
radio tx 5cfea21d0
radio tx 5cfea21f1
radio tx 5cfea2212
radio tx 5cfea2243
radio tx 5cfea2264
radio tx 5cfea2285
radio tx 5cfea22a6
radio tx 5cfea22d7
radio tx 5cfea22f8
radio tx 5cfea2319
radio tx 5cfea234a
radio tx 5cfea236b
radio tx 5cfea238c
`

lolsborn, 2019-06-13:
Could you test out this branch and see if it works for you? https://github.com/doismellburning/LoStik/tree/feature/LineReader-send-example

> I was able to reproduce this issue this morning. It seems to only affect the EU version for some reason, not the US radios. I'm guessing more configuration is required on startup, but it has been tested with earlier versions of this code so I'm not sure what is going on. I'll continue to investigate and hopefully push a patch today.
> 
>> Hello, beginning of the year I bought a pair of 868MHz LoStik and now I'm trying to have the examples working.

>> It turns out that the "radio_receiver.py" doesn't receive any data. I modified it to print every call to "handle_line" and here is the output I get :

>> `(LoEnv) thor@valhalla:~/Geeking/LoStik/examples$ ./radio_receiver.py /dev/ttyUSB1
connection made
handle_line(data='RN2483 1.0.4 Oct 12 2017 14:59:25')
RN2483 1.0.4 Oct 12 2017 14:59:25
handle_line(data='4294967245')
4294967245
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='busy')
handle_line(data='ok')
handle_line(data='ok')
handle_line(data='busy')
handle_line(data='radio_err')
handle_line(data='ok')
handle_line(data='radio_err')
handle_line(data='ok')
... and so on...
`
>> The sender seems to work be working fine :

>> `(LoEnv) thor@valhalla:~/Geeking/LoStik/examples$ ./radio_sender.py /dev/ttyUSB0
radio tx 5cfea21d0
radio tx 5cfea21f1
radio tx 5cfea2212
radio tx 5cfea2243
radio tx 5cfea2264
radio tx 5cfea2285
radio tx 5cfea22a6
radio tx 5cfea22d7
radio tx 5cfea22f8
radio tx 5cfea2319
radio tx 5cfea234a
radio tx 5cfea236b
radio tx 5cfea238c
`

lolsborn, 2019-06-14:
This should be fixed in b12f1be I merged the fork that I mentioned previously and it seems to work.

closed 2019-06-14 by lolsborn

m0nkeyc0de, 2019-06-14:
That was fast and efficient ! Now it works perfectly :+1: 
Many thanks for your help