ronoth/LoStik #19: Lora Class C Example

gsvitak, 2019-06-13:
Hello

Do you have a sample or a suggestion on how to set the device to be a Class C Device?

Thanks
Greg

lolsborn, 2019-06-14:
If you take a look at the latest command reference http://ww1.microchip.com/downloads/en/DeviceDoc/RN2903%20LoRa%20Technology%20Module%20Command%20Reference%20User%20Guide-DS40001811B.pdf

There is a command ```mac set class c``` that should do the trick, but you'll have to first upgrade to 1.0.5 firmware using the LoRa Development Suite GUI https://www.microchip.com/developmenttools/ProductDetails/dv164140-1

closed 2019-06-14 by lolsborn

gsvitak, 2019-06-16:
@lolsborn I cannot find the place to download firmware version 1.0.5.

lolsborn, 2019-06-16:
Are you using the RN2903 (US) or RN2843 (EU)?

Here is the US version link:
https://www.microchip.com/wwwproducts/en/RN2903

If you go to *software *you'll want to download two things

1) LoRa Development Suite (You'll use this GUI to flash the device)
2) RN2903 Firmware v1.0.5 <http://ww1.microchip.com/downloads/en/DeviceDoc/RN2903_1.0.5_Nov_06_2018_10.45.27.zip>

There are two copies of the firmware one is the "unified" version which includes the bootloader which is used for flashing using a Pic programmer. You want the other one, that one doesn't contain the bootloader and is used for programming over USB.

camilo-17, 2019-12-10:
Hello excuse me english I want to know if everyone knows how read the value that sends the LoRa server in the node. I configure the RN2903a to class c and it responds ok, in class a it works correctly, Thanks.