ronoth/LoStik #20: How to use the miniterm?

llorllale, 2019-07-06:
I'm having trouble using `miniterm` to send commands to the LoStik. I'm on Ubuntu 16.04 and using `xterm-256color`.

`miniterm` seems to run fine - I just can't make it execute any of the commands I input. I expect that when I press the Enter key, the command would be sent to the LoStik and I would receive a response. Am I doing something wrong?

Here's an example of what I mean. Notice the cursor about 4 lines below the command - that's me pressing the Enter key several times:

[image]

llorllale, 2019-07-06:
FYI `blinky.py` runs fine

lolsborn, 2019-07-08:
Sorry for the late reply.  The command you are sending is correct.  Can you try sending `sys get ver`?  The command you are sending should still reply with "OK", but get ver at least returns something more substantial.

I have had one other support request recently where a guy was running things just fine, but having issues with mini-term and then plugged it into a different USB port or something and then it worked fine and continued working when plugged in the old port.  I don't know if it will fix your issue here as I haven't quite figured out what causes it and it seems kind of random.  My best guess is a driver issue in the CH340 code.

lolsborn, 2019-07-18:
Would love to know if you were able to work this out.  For now I'm going to close this ticket, but please re-open if you have additional info.  If you suspect it is faulty hardware send an email to steven@ronoth.com

closed 2019-07-18 by lolsborn

LalaDK, 2020-02-04:
I had similar problems with miniterm.py. I use Terminator as my default terminal emulator, but that didn't work. Then I tried xterm and then everything worked! 
Hope that helps !

lolsborn, 2020-02-04:
Terminator works well, but you have to send \r\n line endings

gudsi, 2020-04-17:
Hi!
I have the same issue as well. 
I am trying to use miniterm via cygwin to send commands to lostik which is plugged to my raspberry. Connect Script works fine, but manual commands to not work. 
Can you help me?

lolsborn, 2020-04-17:
What are you using to connect manually?

gudsi, 2020-04-18:
[image]

Pressing enter, but nothing happens

lolsborn, 2020-04-18:
> image <https://user-images.githubusercontent.com/34402250/79632138-aba6d300-815d-11ea-9b91-9b1e2c215ef3.png>

This looks like it's potentially a hardware failure mode that I've seen in the past. Could you email steven@ronoth.com and we can see what to do about it and if we're going to need to mail out a replacement.

gudsi, 2020-04-19:
Thanks for your quick answer. 
The funny thing, I did a reboot and now it works. But my stick is only sending with SF12. Could this issue be related to that hardware topic?

lolsborn, 2020-04-19:
Generally, if the device turns on and replies to commands it is fine. I've seen 2-3 other devices (maybe 1 per 2,000) which enumerate over USB, but never print anything on the console. It's more common that people are using Putty or some terminal that only sends \n and not \r\n so I have to rule that out, but since you're using the mini-term included I was leaning toward that. So if you are sending commands and getting replies it's probably not a hardware problem but a configuration issue.

You say it's only sending on SF12? So when you try to change the spread factor it's not working? Are you running the radio send/receive examples or trying to connect over LoRaWAN?

gudsi, 2020-04-19:
Yes. Sending commands and getting replies is working fine now. 

I am able to adapt the SF via miniterm. If I send data afterwards, I see them arriving, but with SF 12. 

Here is a screenshot of the commands I am sending. 
[image]

I am not working with the sender and receiver examples. I am conntected to a LoRa network

lolsborn, 2020-04-19:
> Here is a screenshot of the commands I am sending. image <https://user-images.githubusercontent.com/34402250/79681530-4c0efd00-821b-11ea-9f18-6bccbc123377.png>

Huh, that is pretty curious. What kind of gateway are you using?

gudsi, 2020-04-19:
CISCO IXM-800 is in use.

gudsi, 2020-04-20:
Does the Stick maybe adapt the SF if the signal is bad?

lolsborn, 2020-04-20:
> Does the Stick maybe adapt the SF if the connection is bad?

I haven't had time to look into this really closely. I'm pretty sure the module will change to SF12 during OTAA negotiation. It's possible that it is stomping on whatever setting you added manually during OTAA, but based on your comments earlier it's staying in SF12 even when you change it after you are connected?

gudsi, 2020-04-21:
Yes. I changed it, after I was connected. 
So first, OTAA, next sending payloads. As mentioned, they are only sent with SF12. No matter which SF is configured via Miniterm. 
I hope, you can help me. I am really stuck

lolsborn, 2020-04-23:
Sorry for the delay. Unfortunately, I don't have an answer for you yet. I'm posting on the Microchip forums and might do the same on the TTN forums and see if anyone has run into a similar issue with the RN2903 / RN2843. I'll keep you updated.

gudsi, 2020-04-27:
Good news. Finally I am sending with a better SF. 
I am using ADR and started to send every 20 minutes. SF is getting better and is now at SF9 or 8.

lolsborn, 2020-04-27:
This is great news. I'm interested in what you did to fix it. You just turned ADR on?