ronoth/LoStik #24: Appkey invalid in lorawan_sender.py

gudsi, 2019-12-31:
Hi!

I am trying to connect my LoStik to The Things network. I tried to run lorawa_sender.py
(https://github.com/ronoth/LoStik/blob/4b109759486caca33381d06ae525d0ba2a6854f1/examples/lorawan_sender.py)

I always get the error message 
 error: argument --appkey/-a: invalid <lambda> value

Can someone help me to solve that?

Thanks a lot

lolsborn, 2019-12-31:
>  error: argument --appkey/-a: invalid value

Can you share what you are running via the command line?

There is also a script called configure.py which will read the settings in ttn-us.conf or ttn-eu.conf and configure the channels/credentials on your device for The Things Network. All you have to do is edit the conf file to suit your needs and run the configure.py script.

gudsi, 2020-01-01:
Thanks a lot for your quick answer! 

I am trying to run it via Pycharm with this Run Parameters:
--payload 1234 --appkey 10DE087B0D8E71B28CD7409B372A2606 --deveui 0004A30B002374AB --appeui 70B3D57ED0027B03 COM3

I also tried to run the configuration first. After adapting the appkey, deveui and appeui in ttn-eu.conf, running configure.py raises an exception
configparser.NoOptionError: No option 'appeui' in section: 'otaa'

lolsborn, 2020-01-02:
The ttn-eu.conf looks like it's not as complete as the US version. If you look at the US conf you can copy and paste the bits you need for OTAA into the eu config and update them with your credentials.

lolsborn, 2020-01-02:
I'm not sure why running the lorawan_sender script you linked would return invalid appkey unless you had the wrong credential.  Is there any reason you are using that script instead of the one in master https://github.com/ronoth/LoStik/blob/master/examples/lorawan.py ? I know that script worked in the past, but the one in master has been tested far more.

gudsi, 2020-01-03:
I added the other channels to my .conf like that
ch16 = , off
ch17 = , off
ch18 = , off 
.... 
ch71 = , off

But still there is the error _configparser.NoOptionError: No option 'appeui' in section: 'otaa'_

I also tried to run the lorawan.py which gives following output
_Connection to LoStik established
mac set appeui XXX
mac set appkey XXX
mac set deveui XXX
mac join otaa
STATUS: ok
STATUS: ok
STATUS: ok
STATUS: ok
STATUS: denied_

I am sorry, but I really do not know, what is going wrong here. 

> The ttn-eu.conf looks like it's not as complete as the US version. If you look at the US conf you can copy and paste the bits you need for OTAA into the eu config and update them with your credentials.
> […](#)

lolsborn, 2020-01-03:
What gateway are you using?

gudsi, 2020-01-03:
Sorry for the stupid answer, but I don´t know. How can I configure that?

lolsborn, 2020-01-03:
LoRaWAN is an authentication, addressing, and management layer on top of LoRa radio that requires a gateway to manage. If you don't have something that resembles a router to receive your LoRa messages you probably don't need LoRaWAN.

You can still use LoRa radios to form simple single channel, point-to-point networks. The other send and receive examples included do that.

gudsi, 2020-01-03:
Thanks for your answer. 
I think, I am a bit confused. 
I want to build a moisture sensor application with RaspberryPi and using the LoStik to send the data. I thought I can use TTN to send data there. Did I get it wrong?

lolsborn, 2020-01-03:
So you are connecting LoStik to RPI and the moisture sensor speaks LoRa and relays it back?

gudsi, 2020-01-03:
Yes. The sensor collects data and sends it via LoRa payload to TTN.

lolsborn, 2020-01-03:
could you share what sensor you are using?

gudsi, 2020-01-04:
I will use this one

gudsi, 2020-03-30:
An update from my side
I finally was able to connect with gateway via Termite and lorawan.py via my pc. But I use LoStik on Raspberry, I get denied or no_channel_free. 
Any ideas what I am doing wrong?

lolsborn, 2020-03-31:
This is usually a channel configuration issue. You have to configure the channel list on your LoStik to match your gateways frequency plan. There is a configure.py script I created to make this easier. You pass it one of the .conf files (after editing it to match your needs). The example ttn.conf files should configure your channels to the ttn frequency plans.

gudsi, 2020-04-08:
Thanks a lot for your help, lolsborn!
I did a big mistake. I did not send a new join after powering the stick again. 
Now everything is working fine. Thanks!

closed 2020-04-08 by gudsi