ronoth/LoStik #26: Receiver example not working with Arduino MKR WAN 1310

rrbonham96, 2020-04-29:
I'm trying to send basic Lora packets to the LoStik using the radio receiver example using an Arduino running the example radio sender code from [https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaSender/LoRaSender.ino](url), but I'm not seeing any data show up on the receiving end. I'm not really sure what I'm doing wrong.

Arduino serial monitor:
`14:34:36.035 -> LoRa Sender
14:34:36.461 -> Sending packet: 0
14:34:41.505 -> Sending packet: 1
14:34:46.534 -> Sending packet: 2
14:34:51.560 -> Sending packet: 3
14:34:56.607 -> Sending packet: 4
14:35:01.643 -> Sending packet: 5
...
`

radio_receiver.py output:
`connection made
RN2903 1.0.5 Nov 06 2018 10:45:27
4294967245
`

lolsborn, 2020-04-30:
> I'm trying to send basic Lora packets to the LoStik using the radio receiver example using an Arduino running the example radio sender code from https://github.com/sandeepmistry/arduino-LoRa/blob/master/examples/LoRaSender/LoRaSender.ino <https://www.fastmail.com/mail/compose/url>, but I'm not seeing any data show up on the receiving end. I'm not really sure what I'm doing wrong.

Based on that example code there seem to be a lot of things missing. Do you need to specify which pins the SPI is hooked up to on the LoRa module? Spi frequency? Set the coding rate? Spread factor?

rrbonham96, 2020-08-31:
I did some research and finally found the solution to this. The issue wasn't anything to do with SPI since the LoRa module is built into the MKR WAN 1310. I had to set the spread factor on the LoStik to 7 since that's what the Arduino defaults to. I also had to set the sync word to 12 to match the default used by the Arduino. I _think_ that's all it took to get communication up. I am still getting radio_errors for the actual data, but that's a different issue I have yet to look into.

closed 2020-08-31 by rrbonham96