ronoth/LoStik #29: how to recover bricked RN2903 with ICSP?

haimiko, 2021-12-08:
Midway into the firmware update my device lost connection.  The module is now bricked.  How can I leverage the ICSP terminals to recover it like with an Arduino as ISP?

lolsborn, 2021-12-08:
It is a PIC microcontroller and you need a programmer for it.  You can get knock-offs cheap https://www.amazon.com/NOYITO-Microcontrollers-PICkit2-Debugger-Programmer/dp/B07GMH732N/ref=sr_1_6?keywords=pickit2&qid=1638996760&sr=8-6

Microchip provides two separate hex files one is "combined" which includes the bootloader.  If you are flashing via USB you want the other one, but over ICSP you want to use the file which includes bootloader.

lolsborn, 2021-12-08:
I guess I should mention you'll want to download MPLab IDE for this.

haimiko, 2021-12-08:
@lolsborn figured it out without the programmer.  Since I bricked it using the USB (bootloader) method the bootloader was still intact.  I found an alternate solution using the Microchip Suite --> Module --> Boot Loader Recover, then loaded the hex file successfully.

Your method would definitely come in handy should I corrupt the bootloader too.  Thank you.

closed 2021-12-08 by haimiko