ronoth/LoStik #32: Status mac paused

sainathgorige, 2023-07-25:
Hello,

I'm trying to get coverage of my lora network using losik, I bought two losik, I have tried using lorawan.py, but couldn't succeed.

Following are the steps I followed:

    Generated random unique values(also used sensor values) for DevEUI, AppEUI, and AppKey.
    Registered these values with your LoRaWAN network server(chirpstack dashboard).
    Use these same values in your Python script when running it with the LoStik device.

but the script hung at status ,ac paused:
 sudo python3 lorawan.py /dev/ttyUSB0 --appeui 70B3D57ED0031D3D --appkey EC925802AE430F59798E8C8D7E926C8D --deveui 00D827F000B4C638
Connection to LoStik established
mac set appeui 70B3D57ED0031D3D
mac set appkey EC925802AE430F59798E8C8D7E926C8D
mac set deveui 00D827F000B4C638
mac join otaa
STATUS: ok
STATUS: ok
STATUS: ok
STATUS: mac_paused

is there anything I'm missing

I have also tried using ABP

sudo python3 lorawan.py /dev/ttyUSB0 --joinmode abp --devaddr 010bb971 --nwkskey c44e91483a00aa3cedba46bc4ebe87a8 --appskey d7321ca80503973775a521759f3c5da7
[sudo] password for aerpaw:
Connection to LoStik established
mac set devaddr 010bb971
mac set appskey d7321ca80503973775a521759f3c5da7
mac set nwkskey c44e91483a00aa3cedba46bc4ebe87a8
mac join abp

lolsborn, 2023-07-25:
> @lolsborn <https://github.com/lolsborn> @doismellburning
> <https://github.com/doismellburning> @eggfriedrice
> <https://github.com/eggfriedrice> any help?

Did you configure the channels?  What network are you trying to connect to
and what region are you in?

sainathgorige, 2023-07-25:
No, How to configure the channels, can you please give a doc/example?

I have 6 RAK lora gateways, which constitutes my lora network. and I'm using US 915mhz.

lolsborn, 2023-07-25:
Take a look at the configure.py script in the examples directory.  It
expects a configuration script like the ttn-us.conf file which is setup for
"The Things Network" with the US channel plan.  You'll need to run that or
manually send the channel commands over to configure them.

sainathgorige, 2023-07-25:
Thanks, I have added deveui and app creds to the ttn-us-conf, But still I'm facing the same issue
I have attached the logs to this comment.

[ttn-us.txt](https://github.com/ronoth/LoStik/files/12165835/ttn-us.txt)

[losik_logs.txt](https://github.com/ronoth/LoStik/files/12165829/losik_logs.txt)